kevinmehall/nusb #4: Timeouts for blocking IO

kevinmehall, 2023-10-29:
For Rust async, it's normal for futures not to have their own timeouts. To implement a timeout you race the future with a timeout future such that the main future will be cancelled if the timeout completes first. But this requires a reactor to provide timer functionality as a generic future, and I want to make `nusb` not require pulling in `tokio` or `async-std` for the common case where you don't actually need async in the rest of the app.

`futures_lite::future::block_on` is implemented on top of [parking](https://docs.rs/parking/2.2.0/parking/struct.Parker.html), which has an API with a timeout. So we could see if `futures_lite` would add a timeout (seems like many uses of blocking on async IO would want this?), find another implementation that has this to recommend, or copy that code here.

XVilka, 2024-06-24:
It is definitely useful, given that `libusb-rs` provided this feature for transactions.

wuwbobo2021, 2024-11-10:
Inspired by <https://docs.rs/probe-rs/latest/src/probe_rs/probe/usb_util.rs.html> (which probably discards possible data on timeout), I wrote a prototype (as a draft) of synchronous reader for `nusb`, but I'm not sure if it is the correct implementation. And I can't unify synchronous read and write operations into a wrapper of `Queue` externally, because the generic type restriction `platform::TransferData: PlatformSubmit<R>` is held privately inside `nusb`.

```rust
type ReadQueue = nusb::transfer::Queue<nusb::transfer::RequestBuffer>;
type WriteQueue = nusb::transfer::Queue<Vec<u8>>;

pub(crate) struct BulkReader {
    queue: ReadQueue,
    buf: Option<Vec<u8>>,
    timeout: Duration,
}

impl BulkReader {
    fn new(queue: ReadQueue, timeout: Duration) -> Self {
        Self {
            queue,
            buf: Some(Vec::new()),
            timeout,
        }
    }
}

use futures_lite::FutureExt;
impl Read for BulkReader {
    fn read(&mut self, buf: &mut [u8]) -> std::io::Result<usize> {
        use nusb::transfer::TransferError;
        let buf_async = self.buf.take().unwrap();
        let req = nusb::transfer::RequestBuffer::reuse(buf_async, buf.len());
        self.queue.submit(req);
        let fut = self.queue.next_complete();
        let fut_comp = async { Some(fut.await) };
        let fut_cancel = async {
            async_io::Timer::after(self.timeout).await;
            None
        };
        let comp = {
            let mut maybe_comp = async_io::block_on(fut_comp.or(fut_cancel));
            if maybe_comp.is_none() {
                self.queue.cancel_all(); // the only one
                let comp = async_io::block_on(self.queue.next_complete());
                maybe_comp.replace(comp);
            }
            maybe_comp.unwrap()
        };
        let result = match comp.status {
            Ok(()) => {
                // 
                buf.copy_from_slice(&comp.data);
                Ok(comp.data.len())
            }
            Err(TransferError::Cancelled) => {
                if !comp.data.is_empty() {
                    buf.copy_from_slice(&comp.data);
                    Ok(comp.data.len())
                } else {
                    Err(Error::from(ErrorKind::TimedOut))
                }
            }
            Err(TransferError::Disconnected) => Err(Error::from(ErrorKind::NotConnected)),
            Err(TransferError::Stall) => {
                self.queue.clear_halt(); // is copy_from_slice() needed?
                Err(Error::other(TransferError::Stall))
            }
            Err(e) => Err(Error::other(e)),
        };
        self.buf.replace(comp.data);
        result
    }
}
```

If `copy_from_slice()` should be avoided, I'll probably do so:
```rust
// replaces `let buf_async = self.buf.take().unwrap();`
let buf_async = unsafe { Vec::from_raw_parts(buf.as_mut_ptr(), buf.len(), buf.len()) };
// ...
// replaces `self.buf.replace(comp.data);`
let _ = Box::into_raw(comp.data.into_boxed_slice());
```
I think it's highly unsafe, do you think it's possible?

wuwbobo2021, 2024-11-11:
Sorry, I've reviewed my previous code and found a bad mistake: `copy_from_slice()` will panic because it was used wrongly, so I hid it as `spam` for a while. The current code is tested by myself.

Will this feature be added any time soon?

wuwbobo2021, 2024-12-01:
I've changed my code again to use the `futures_timer` crate, it doesn't introduce any extra dependency by default.

kevinmehall, 2025-05-04:
https://github.com/kevinmehall/nusb/pull/117 adds `Endpoint::wait_next_complete` which blocks with a timeout, will be in v0.2.

closed 2025-05-04 by kevinmehall