kevinmehall/nusb #23: class, subclass, and protocol return zero

lethargy123, 2024-01-05:
The `class`, `subclass`, and `protocol` values returned by the `list_devices` function are all zero, which is considered abnormal. However, when using `libsub` to read the class, subclass, and protocol of the winus device, the values are 0xFF, 0x42, and 0x03 respectively, which is the expected result.
[image]

kevinmehall, 2024-01-06:
Hmm. On Windows, these are [coming straight from](https://github.com/kevinmehall/nusb/blob/54d3eee07dd0362740f3a02cd26a2e19d860f528/src/platform/windows_winusb/enumeration.rs#L84-L86) a call to  [IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/usbioctl/ni-usbioctl-ioctl_usb_get_node_connection_information_ex). I believe libusb also uses the same call to get the device descriptor, so I'll have to see if I can reproduce this and see what we're doing differently.

kevinmehall, 2024-01-10:
I'm not able to reproduce this. I tested with a few Android phones in normal mode and fastboot mode, and on all of them the class, subclass, and protocol read as zero in nusb and libusb on both Windows and Linux, but this is the correct value from the device descriptor. Zero is the typical value for a composite device and means that the class is defined at the interface level.

However, Android fastboot does return an **interface descriptor** with class=0xFF, subclass=0x42, and protocol=3. Some Windows tooling, including the Device Manager under Compatible IDs, shows these values for the whole device. Are you sure you are expecting to see these values in the device descriptor, and not the interface descriptor? `nusb` correctly reads these values from the interface descriptor (see the `descriptors` example).

kevinmehall, 2024-01-15:
https://github.com/kevinmehall/nusb/pull/29 adds `DeviceInfo::interfaces` to get class, subclass, protocol on each interface, as long as the device is using the composite group parent driver.

closed 2024-01-15 by kevinmehall