kevinmehall/nusb #28: API to get interface string, cached if possible.

Dirbaio, 2024-01-09:
The use case is a bit cursed, the CMSIS-DAP debug probe spec [says](https://arm-software.github.io/CMSIS_5/DAP/html/group__DAP__ConfigUSB__gr.html) that to enumerate CMSIS-DAP devices you have to look for devices where the product string, or for composite devices the interface string, contain "CMSIS-DAP"

> Update String Settings - Product String to indicate the debug unit. Note that "CMSIS-DAP" must be part of that string to allow identification by debuggers (or part of interface string for USB composite device).

This is the worst case possible, to find CMSIS-DAP devices we have to read all interface strings for all USB devices in the system! :( 

I've found Windows caches it as `DEVPKEY_Device_BusReportedDeviceDesc` for the interface device nodes (the ones with `&MI_00` in the path), Linux caches it in sysfs (see [here](https://github.com/probe-rs/probe-rs/pull/2032)), no idea about Mac.

similar to [here](https://github.com/kevinmehall/nusb/issues/24#issuecomment-1883490036) it'd be nice to have some API that does all the OS-specific cache stuff for you and reads it from the hardware if not possible.

kevinmehall, 2024-01-10:
Yep, this would be useful. I started implementing something for Linux at https://github.com/kevinmehall/nusb/pull/29. What do you think of that API?

Poking at `ioreg` on macOS, it looks similar to Windows and Linux in that there's a child device node representing the interface.

I think the only case where this info may not be available is on a non-composite device on Windows, because if the entire device is bound to a driver other than usbccgp, there won't be child nodes. But I think the use cases are mostly around composite devices so maybe that is not a big deal.

kevinmehall, 2024-01-15:
Implemented in #29

closed 2024-01-15 by kevinmehall