kevinmehall/nusb #33: Atmel SAMD21 device

Resonanz, 2024-01-16:
I have a SAMD21 uC connecting to Windows using a fully functioning custom .NET app using HID.

I want to move over from C# to Rust.

I have the VID and PID:

`.find(|d| d.vendor_id() == 0x04d8 && d.product_id() == 0xeb88)`

If I `cargo run --example control` I get:

```
     Running `target\debug\examples\control.exe`
Device info: DeviceInfo { bus_number: 0, device_address: 21, vendor_id: 0x04D8, product_id: 0xEB88, 
device_version: 0x0100, class: 0xEF, 
subclass: 0x02, protocol: 0x01, speed: Some(Full), manufacturer_string: None, product_string: Some("Device"), 
serial_number: Some("FFFF"), instance_id: "USB\\VID_04D8&PID_EB88\\FFFF", 
parent_instance_id: "USB\\VID_2109&PID_2813\\6&19d7c37a&0&3", port_number: 1, 
driver: Some("usbccgp"), interfaces: [InterfaceInfo { interface_number: 0, class: 0x02, subclass: 0x02, 
protocol: 0x01, interface_string: Some("Device") }, InterfaceInfo { interface_number: 2, 
class: 0x03, subclass: 0x00, protocol: 0x00, interface_string: Some("Device") }] }

thread 'main' panicked at examples\control.rs:40:47:

called `Result::unwrap()` on an `Err` value: Custom { kind: NotFound, error: "interface not found or not compatible" }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
error: process didn't exit successfully: `target\debug\examples\control.exe` (exit code: 101)
```

I have traced the failure to `pub(crate) fn claim_interface` (line 107) in file `\src\platform\windows_winusb\device.rs`

The value of `interface: u8` is `"\0"`, e.g. NULL

```
       let path = self.interface_paths.get(&interface).ok_or_else(|| {
            Error::new(ErrorKind::NotFound, "interface not found or not compatible")
```

Any thoughts on this? The device is being identified as follows:

[image]

Resonanz, 2024-01-16:
On Ubuntu:

```
Device info: DeviceInfo { bus_number: 2, device_address: 37, vendor_id: 0x04D8, product_id: 0xEB88, 
device_version: 0x0100, class: 0xEF, 
subclass: 0x02, protocol: 0x01, speed: Some(Full), manufacturer_string: Some("Device"), product_string: Some("Device"), 
serial_number: Some("FFFF"), sysfs_path: SysfsPath("/sys/bus/usb/devices/2-2"), 
interfaces: [InterfaceInfo { interface_number: 0, class: 0x02, subclass: 0x02, protocol: 0x01, interface_string: None },
InterfaceInfo { interface_number: 1, class: 0x0A, subclass: 0x00, protocol: 0x00, interface_string: None }, 
InterfaceInfo { interface_number: 2, class: 0x03, subclass: 0x00, protocol: 0x00, interface_string: None }] }
Completion { data: ResponseBuffer { transferred: 0, .. }, status: Err(Stall) }
Completion { data: [], status: Err(Stall) }
thread 'main' panicked at examples/control.rs:40:47:
called `Result::unwrap()` on an `Err` 
value: Os { code: 16, kind: ResourceBusy, message: "Device or resource busy" }
```

kevinmehall, 2024-01-16:
It looks like your device is composite HID and CDC-serial based on the `class` fields. Which of these interfaces are you trying to use? These will have kernel CDC and HID drivers that will prevent use with `nusb`.

What does the device do? If your device is HID and CDC because these are the appropriate classes for the device's functionality (e.g. a Human Interface Device like mouse/keyboard/gamepad for HID, and a serial port you want to use with a terminal emulator for CDC), then you should not use `nusb`, and instead use libraries like `hidapi` and `serialport-rs` to use the kernel drivers to access this functionality through those existing drivers.

Historically, HID was abused for vendor-specific non-HID functionality because you can tunnel a little bit of custom data over HID reports. Windows would install the HID driver automatically, but in the WinXP days before WinUSB, it was hard to install and write a vendor-class driver. A lot of microcontroller USB examples use CDC, because you can get instant gratification by writing to it with `printf`-style APIs and interacting with it in a terminal emulator. If there is not a reason to use the standard classes, consider changing the firmware to replace one or both of these interfaces with a vendor-class (0xFF) interface. Use a Microsoft OS descriptor to set a [Compatible ID](https://github.com/pbatard/libwdi/wiki/WCID-Devices) of `WINUSB` with a random `DeviceInterfaceGUIDs` to make Windows bind the WinUSB driver to that interface. Then you can set up whatever endpoints and control requests you want under that interface, and access them with `nusb`.

On Windows, that's the entire issue -- `nusb` requires that the interfaces have the `WinUSB` driver bound, not the CDC or HID drivers.

On Linux, you are able to make control requests without claiming an interface. It looks like you modified the control example which makes a few control requests using `Device` without claiming the interface. These first transfers fail with `Stall`. This means the request was rejected by the device, typically if it's not a request the device understands. Do the parameters to the control transfer methods correspond to request handlers in your device firmware, or did you leave the random ones I had for my device in the example? Then when it tries to claim the interface, it can't because it's already in use by the kernel HID or CDC drivers.

Resonanz, 2024-01-17:
Kevin, thanks.

You are correct that this is a composite device. It was originally designed to operate using Christian Findlay's C# libraries, allowing both to run HID and CDC together.

[https://www.christianfindlay.com/blog/device-net-4](https://www.christianfindlay.com/blog/device-net-4)

You are also correct that the purpose of this setup was to provide seamless plug-and-play operation as well as text based serial port commands. This is highly desirable for the product.

Thanks a lot for the detailed reply -- I have read over it multiple times and will keep doing so as there is a lot in there. Meanwhile I will kee a close eye on your project and look into "vendor-class".

closed 2024-01-17 by kevinmehall