kevinmehall/nusb #42: Get device address

boozook, 2024-02-27:
Could be great to have something like this in libusb:

> `uint8_t libusb_get_device_address(libusb_device *dev)` Returns the device_address contained by the device dev. ([src](https://nxmnpg.lemoda.net/3/libusb_get_device_address))

kevinmehall, 2024-02-27:
There's already [`DeviceInfo::device_address`](https://docs.rs/nusb/latest/nusb/struct.DeviceInfo.html#method.device_address).

closed 2024-02-27 by kevinmehall

boozook, 2024-02-27:
That's cool, thanks and sorry for disturb, @kevinmehall.

Also if s there any semi-crossplatform method to determine mount point of a mass-storage device that automatically mounted by OS?

kevinmehall, 2024-02-28:
No worries.

> Also if s there any semi-crossplatform method to determine mount point of a mass-storage device that automatically mounted by OS?

That's a bit out of scope for nusb, and I haven't looked if there are other libraries for that.