kevinmehall/nusb #47: Support for isochronous transfers

alufers, 2024-03-05:
Hi!

I am in progress of creating a [Rust application](https://github.com/alufers/thermal-cat) to display a feed from thermal cameras. They are UVC devices (with some nasty non standard features, which require an userspace driver), which use isochronous transfers to send the video.

It would be very nice to have a native Rust API for receiving isochronous data, since rusb doesn't wrap libusb for that and playing around with unsafe code, callbacks, and such is required. 

PS: If you think this is out of scope to this project, feel free to close this issue. Many thanks for your library!

kevinmehall, 2024-03-07:
This is definitely in scope, but not something I'm going to be able to prioritize in the near future.

If you (or someone else reading this) is interested in working on this, the first step is to figure out what the API should be. I think it probably makes most sense to build on `Queue` as isochronous is generally used for streaming and requires multiple submitted transfers to not lose packets. [MacOS](https://opensource.apple.com/source/IOUSBFamily/IOUSBFamily-198.4.15/Documentation/IOUSBLib/Functions/InterfaceInterface.html#WriteIsochPipeAsync) and [Linux](https://github.com/torvalds/linux/blob/master/tools/include/uapi/linux/usbdevice_fs.h#L111C8-L115) have per-packet metadata that must be dealt with when allocating and submitting transfers. [Windows](https://learn.microsoft.com/en-us/windows/win32/api/winusb/nf-winusb-winusb_readisochpipeasap) has this for IN transfers only, but requires that OUT transfers are contiguous in the buffer. Windows also requires a mechanism for pre-registering buffers (maybe like #11?).

andreysaf, 2024-07-15:
Would love to have support for isochronous transfer. I am interested in web route and seeing it added it to [cross-usb](https://crates.io/crates/cross_usb) later.

Great library indeed! Thank you.

NuLL3rr0r, 2025-03-07:
+1 this will be a nice addition. This was the reason we looked into nusb I see that pull request #117  contains commit with isochronous support. We'd appreciate it if you'd let us know if it's functional at this stage, so we could apply the patch.

Thank you!

kevinmehall, 2025-03-07:
#117 is designed with isochronous in mind, but does not implement isochronous yet.

NuLL3rr0r, 2025-03-07:
Thanks for the quick reply! Looking forward to it.

JRazek, 2025-07-03:
Is there any progress on this?

kevinmehall, 2025-07-03:
Not yet.

Jobians, 2026-09-21:
Does this project support uvc?

kevinmehall, 2026-09-21:
Depends on your device. UVC can be either bulk or isochronous endpoints. But unless you're doing something particularly special, you're probably better off using kernel UVC drivers rather than implementing UVC in user space on top of nusb.

Jobians, 2026-09-21:
Thanks for the clarification. I’m specifically trying to get an MPEG stream from a USB video capture device that uses UVC.

I know "libuvc-sys" can handle this, but I’m wondering whether this pure-Rust library currently supports UVC devices directly, including the relevant bulk/isochronous video endpoints, or whether UVC is outside the scope of the project.

kevinmehall, 2026-09-21:
nusb is lower level and implementation of any class is out of scope. You want something that uses the kernel's UVC drivers, not generic USB access.