kevinmehall/nusb #48: WinUSB: Allow claiming additional interfaces with `WinUsb_GetAssociatedInterface`

kevinmehall, 2024-03-14:
We currently only use [`WinUsb_Initialize`](https://learn.microsoft.com/en-us/windows/desktop/api/winusb/nf-winusb-winusb_initialize) to obtain a WinUSB handle. When multiple USB interfaces are represented by a single node in the Windows driver model, `WinUSB_Initialize` opens the first such interface, and then [`WinUsb_GetAssociatedInterface`](https://learn.microsoft.com/en-us/windows/win32/api/winusb/nf-winusb-winusb_getassociatedinterface) must be used to open the others.

There are two cases where this occurs:
* When interfaces are grouped with an interface association descriptor (IAD)
* When an entire composite device is bound to WinUSB, instead of usbccgp creating per-interface device nodes.
   * Prior to https://github.com/kevinmehall/nusb/pull/41, nusb would fail to find such interfaces and pretend that only interface 0 exists. Now it ignores the interface number and tries to open the first interface no matter which interface number is passed.

bugadani, 2024-05-03:
I have seen cases in Windows where nusb lists 0 interfaces for otherwise working devices. Using https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/usbview#usbview-source-code I can see several interfaces. Are these symptoms related to this issue?

kevinmehall, 2024-05-03:
It's likely related to the second case above. Does the `driver` on `DeviceInfo` show `WINUSB`?

Right now, it [only tries](https://github.com/kevinmehall/nusb/blob/38aee4a698fba3a4e223a841e3e3437b766dc501/src/platform/windows_winusb/enumeration.rs#L62) to list interfaces that are represented by child objects of a composite group parent device. If the whole device is bound to WinUSB, there will only be a single device object that represents the device and all interfaces.

Normally devices that install the driver this way (via MS OS / WCID descriptors or otherwise) are vendor class devices with a single interface, so the interface list isn't that interesting. But it's possible to put a composite device in this state with Zadig.

It probably needs to get the interfaces from the cached config descriptor instead of using cfgmgr32 to list device objects. That would find the same interfaces as `dev.active_configuration()?.interfaces()` after opening the device.

PR #57 (2024-07-28): windows: Allow claiming interfaces that require WinUsb_GetAssociatedInterface
Fixes #48
closed 2024-07-28 by kevinmehall