kevinmehall/nusb #59: Async methods to get descriptors

axelkar, 2024-06-11:
Currently `get_descriptor`, `get_string_descriptor` and `get_string_descriptor_supported_languages` all block. Could async versions of these methods be made?

kevinmehall, 2024-06-11:
On macOS and Linux, `get_descriptor` just calls `control_in_blocking` and could just as easily use the async `control_in`, or you could replicate it with `control_in` externally. On Windows, it calls [IOCTL_USB_GET_DESCRIPTOR_FROM_NODE_CONNECTION](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/usbioctl/ni-usbioctl-ioctl_usb_get_descriptor_from_node_connection). Windows `DeviceIoControl` takes an `overlapped` parameter that potentially allows it to be used with IOCP asynchronously, but I haven't tested this.

kevinmehall, 2025-05-04:
Implemented in https://github.com/kevinmehall/nusb/pull/117 -- these now return `MaybeFuture` and can be used in blocking or async mode. Implementation on Windows just using the `blocking` threadpool for now.

closed 2025-05-04 by kevinmehall