kevinmehall/nusb #61: `Interface::bulk_out()` - support also slices as an argument

XVilka, 2024-06-24:
Currently, the method signature is 
```
pub fn bulk_out(&self, endpoint: u8, buf: Vec<u8>)
```

But it makes a requirement to allocate the buf, which isn't always welcome. I suggest also to support specifying `buf` as `&[u8]` slice.

kevinmehall, 2024-06-24:
Unfortunately an API taking a borrowed slice is not safe when it comes to async IO in Rust, because you can leak the Future and regain mutable access to the buffer while the kernel continues to read from it. It's the same problem as [io-uring and other completion-based IO](https://boats.gitlab.io/blog/post/io-uring/).

kevinmehall, 2025-07-27:
v0.2 adds endpoint [adapter types that impl Read and Write](https://docs.rs/nusb/latest/nusb/io/index.html). They copy the data to internally-managed buffers that are used for kernel-level transfers, but those buffers are only allocated once and then re-used.

closed 2025-07-27 by kevinmehall