kevinmehall/nusb #66: `list_devices` does not return devices with large `locationID` value on macOS

tacman09, 2024-07-25:
This issue occurs if `locationID` value is 0x80000000 or more in my case, but I haven't checked the exact boundary value.

The USB devices shown in IORegistryExplorer.
[image]
The output of `examples/list.rs`
```
[2024-07-25T20:50:52Z DEBUG nusb::platform::macos_iokit::enumeration] Probing device 4294985249
[2024-07-25T20:50:52Z DEBUG nusb::platform::macos_iokit::enumeration] Probing device 4294985244
[2024-07-25T20:50:52Z DEBUG nusb::platform::macos_iokit::enumeration] Device does not have property `bInterfaceNumber`
[2024-07-25T20:50:52Z DEBUG nusb::platform::macos_iokit::enumeration] Device does not have property `kUSBString`
[2024-07-25T20:50:52Z DEBUG nusb::platform::macos_iokit::enumeration] Device does not have property `USB Interface Name`
DeviceInfo {
    bus_number: 0,
    device_address: 1,
    vendor_id: 0x0781,
    product_id: 0x5580,
    device_version: 0x0010,
    class: 0x00,
    subclass: 0x00,
    protocol: 0x00,
    speed: Some(
        Super,
    ),
    manufacturer_string: Some(
        "SanDisk",
    ),
    product_string: Some(
        "Extreme",
    ),
    serial_number: Some(
        "AA010425152109400848",
    ),
    location_id: 18874368,
    registry_entry_id: 4294985244,
    interfaces: [
        InterfaceInfo {
            interface_number: 0,
            class: 0x08,
            subclass: 0x06,
            protocol: 0x50,
            interface_string: None,
        },
    ],
}
```

kevinmehall, 2024-07-26:
What macOS version and device model do you have?

Since it is locationID, is it specific to a particular port or set of ports on the machine? If you swap the ports of those two devices, is it the other device that is not listed?

I have to suspect this:

https://github.com/kevinmehall/nusb/blob/bcab2875f07fb1bb5b98c05067f9192a1500108b/src/platform/macos_iokit/enumeration.rs#L130-L134

since if it returns None, it would bail out of listing that device without logging anything else. However it does seem to handle numbers greater than 0x80000000 in testing like this:

```rs
use core_foundation::number::CFNumber;

fn main() {
    let cfn = CFNumber::from(0x80100000i64);
    println!("{cfn:?}");

    let n: u32 = Some(cfn)
        .and_then(|n| n.to_i64())
        .and_then(|n| n.try_into().ok())
        .unwrap();
    println!("{n:x}");
}
```

That  `CFNumber::from` produces a `kCFNumberSInt64Type`. I'm going to take a wild guess that the value from iokit is still a `kCFNumberSInt32Type`, returning a negative number, which then fails in the `i64::try_into::<u32>()`.

Can you try this branch? https://github.com/kevinmehall/nusb/compare/iokit-numbers

Whether it works or not, I'd be interested in the debug log output to confirm that guess.

tacman09, 2024-07-26:
> What macOS version and Apple device model do you have?

macOS Ventura 13.6.7 in M1 MacBook Air 2020

> Since it is locationID, is it specific to a particular port or set of ports on the machine?

The devices with locationID over 0x80000000 appears in the tree when [VirtualHere](https://www.virtualhere.com/) client is running.

> If you swap the ports of those two devices, is it the other device that is not listed?

Yes.

> Can you try this branch? https://github.com/kevinmehall/nusb/compare/iokit-numbers

All devices was successfully listed as below!

```
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Probing device 4294992372
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property locationID
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] value is "<CFNumber 0xfe7fa8487b9527f1 [0x204e8aa90]>{value = -2146435072, type = kCFNumberSInt32Type}"
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property USB Address
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] value is "<CFNumber 0x8180578873952771 [0x204e8aa90]>{value = +1, type = kCFNumberSInt32Type}"
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property idVendor
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] value is "<CFNumber 0x81805788739cd8f1 [0x204e8aa90]>{value = +5118, type = kCFNumberSInt32Type}"
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property idProduct
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] value is "<CFNumber 0x81805788738e3671 [0x204e8aa90]>{value = +13859, type = kCFNumberSInt32Type}"
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property bcdDevice
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] value is "<CFNumber 0x818057887395a7f1 [0x204e8aa90]>{value = +256, type = kCFNumberSInt32Type}"
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property bDeviceClass
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] value is "<CFNumber 0x81805788739527f1 [0x204e8aa90]>{value = +0, type = kCFNumberSInt32Type}"
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property bDeviceSubClass
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] value is "<CFNumber 0x81805788739527f1 [0x204e8aa90]>{value = +0, type = kCFNumberSInt32Type}"
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property bDeviceProtocol
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] value is "<CFNumber 0x81805788739527f1 [0x204e8aa90]>{value = +0, type = kCFNumberSInt32Type}"
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property Device Speed
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] value is "<CFNumber 0x81805788739526f1 [0x204e8aa90]>{value = +2, type = kCFNumberSInt32Type}"
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property USB Vendor Name
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property USB Product Name
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property USB Serial Number
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property bInterfaceNumber
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Device does not have property `bInterfaceNumber`
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property bInterfaceNumber
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] value is "<CFNumber 0x81805788739527f1 [0x204e8aa90]>{value = +0, type = kCFNumberSInt32Type}"
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property bInterfaceClass
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] value is "<CFNumber 0x81805788739523f1 [0x204e8aa90]>{value = +8, type = kCFNumberSInt32Type}"
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property bInterfaceSubClass
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] value is "<CFNumber 0x81805788739524f1 [0x204e8aa90]>{value = +6, type = kCFNumberSInt32Type}"
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property bInterfaceProtocol
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] value is "<CFNumber 0x8180578873950ff1 [0x204e8aa90]>{value = +80, type = kCFNumberSInt32Type}"
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property kUSBString
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Device does not have property `kUSBString`
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Getting property USB Interface Name
[2024-07-26T11:40:44Z DEBUG nusb::platform::macos_iokit::enumeration] Device does not have property `USB Interface Name`
DeviceInfo {
    bus_number: 0,
    device_address: 1,
    vendor_id: 0x13FE,
    product_id: 0x3623,
    device_version: 0x0100,
    class: 0x00,
    subclass: 0x00,
    protocol: 0x00,
    speed: Some(
        High,
    ),
    manufacturer_string: Some(
        "        ",
    ),
    product_string: Some(
        "silicon-power",
[... 61 lines cut ...]
    speed: Some(
        Super,
    ),
    manufacturer_string: Some(
        "SanDisk",
    ),
    product_string: Some(
        "Extreme",
    ),
    serial_number: Some(
        "AA010425152109400848",
    ),
    location_id: 18874368,
    registry_entry_id: 4294992411,
    interfaces: [
        InterfaceInfo {
            interface_number: 0,
            class: 0x08,
            subclass: 0x06,
            protocol: 0x50,
            interface_string: None,
        },
    ],
}
```

closed 2024-07-27 by kevinmehall