kevinmehall/nusb #68: NotFound errors on M2 Macbook Air with Fedora 40 Asahi Remix

i509VCB, 2024-08-02:
From https://github.com/probe-rs/probe-rs/issues/2731 to here. OpenOCD seems unaffected (obviously using libusb)

I noticed while trying to use probe-rs, nusb seems to have problems if `busnum` is missing at some spot in the sysfs.

Log:

```
DEBUG nusb::platform::linux_usbfs::enumeration: probe device SysfsPath("/sys/bus/usb/devices/1-1.1")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/busnum": Ok("1\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/devnum": Ok("5\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/idVendor": Ok("2e8a\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/idProduct": Ok("000c\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/bcdDevice": Ok("0201\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/bDeviceClass": Ok("00\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/bDeviceSubClass": Ok("00\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/bDeviceProtocol": Ok("00\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/speed": Ok("12\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/manufacturer": Ok("Raspberry Pi\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/product": Ok("Debug Probe (CMSIS-DAP)\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/serial": Ok("E6616408432F7133\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/1-1.1:1.2/bInterfaceNumber": Ok("02\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/1-1.1:1.2/bInterfaceClass": Ok("0a\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/1-1.1:1.2/bInterfaceSubClass": Ok("00\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/1-1.1:1.2/bInterfaceProtocol": Ok("00\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/1-1.1:1.2/interface": Err(Os { code: 2, kind: NotFound, message: "No such file or directory" })
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/1-1.1:1.0/bInterfaceNumber": Ok("00\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/1-1.1:1.0/bInterfaceClass": Ok("ff\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/1-1.1:1.0/bInterfaceSubClass": Ok("00\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/1-1.1:1.0/bInterfaceProtocol": Ok("00\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/1-1.1:1.0/interface": Ok("CMSIS-DAP v2 Interface\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/1-1.1:1.1/bInterfaceNumber": Ok("01\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/1-1.1:1.1/bInterfaceClass": Ok("02\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/1-1.1:1.1/bInterfaceSubClass": Ok("02\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/1-1.1:1.1/bInterfaceProtocol": Ok("00\n")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1/1-1.1:1.1/interface": Ok("CDC-ACM UART Interface\n")
DEBUG nusb::platform::linux_usbfs::enumeration: probe device SysfsPath("/sys/bus/usb/devices/1-1.1:1.0")
DEBUG nusb::platform::linux_usbfs::enumeration: sysfs read "/sys/bus/usb/devices/1-1.1:1.0/busnum": Err(Os { code: 2, kind: NotFound, message: "No such file or directory" })
DEBUG nusb::platform::linux_usbfs::enumeration: failed to probe, skipping: No such file or directory (os error 2)
```

Full log:
[nusblog.txt](https://github.com/user-attachments/files/16465001/nusblog.txt)

kevinmehall, 2024-08-02:
The paths with `:` in them are actually USB interfaces, not devices, so failing to read their `busnum` and skipping them is expected (though it could probably ignore them based on the path, rather than trying to read device properties).

The lines
```
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes using nusb, searching HID
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes total
```

and 
```
DEBUG probe_rs::probe::cmsisdap::tools: Attempting to open 2e8a:000c in CMSIS-DAP v1 mode
```

seem to indicate that `nusb::list_devices` successfully returned the device to probe-rs, and the failure happened later. Are you sure this is not a permissions issue or other failure opening the device as suggested by the message?

i509VCB, 2024-08-02:
I doubt the problem is with permissions, as my x86_64 machine required no permission changes. I did add my user on the arm64 machine to the `plugdev` group per some suggestion I found in probe-rs and rebooted. But that didn't fix it.

kevinmehall, 2024-08-04:
https://github.com/kevinmehall/nusb/pull/72 cleans up the benign "No such file or directory" errors and adds logging when opening a device or claiming an interface fails. (It's on probe-rs for ignoring such errors without displaying any information about them, but at least it will be in the nusb log for situations like this).

For further troubleshooting, you'll need to find why opening the device fails, either using the logging now on nusb `main` or by patching probe-rs to do something with the error here: https://github.com/probe-rs/probe-rs/blob/6bcbeca7eb71f92413cf9c3fd3c8b55eb1c942a9/probe-rs/src/probe/cmsisdap/tools.rs#L152

>  I did add my user on the arm64 machine to the plugdev group

I think plugdev is a Debian/Ubuntu thing. It would work to add a `plugdev` group, add your user, and add udev rules that grant it permission, but it's possible you're missing one of those steps. I think the Fedora way is a udev rule with `TAG+="uaccess"` but IIRC there are some rule ordering dependencies that can result in it being silently ignored.

i509VCB, 2024-09-02:
Updating probe-rs seems to have somehow fixed this?

closed 2024-09-02 by i509VCB