kevinmehall/nusb #69: The write speed of the function `bulk_out` is too slow

lethargy123, 2024-08-02:
The write speed of the function `bulk_out` is too slow. This might be because `bulk_out` returns `vec<u8>`. I recommend changing it to return the number of bytes written and then see if the write speed increases. Additionally, I noticed that `bulk_out` does not experience timeouts, which seems somewhat unreasonable to me.

lethargy123, 2024-08-02:
> 当 return 产生的未来时，它只是归还了传递给它的缓冲区的所有权。它不会用发送的数据填充新的缓冲区。无论是什么原因导致了吞吐量问题，返回类型都不是问题。`bulk_out``Vec<u8>`
> 
> 不包括超时是 Rust 期货的标准做法。如果希望操作能够超时，请将其与提供超时的第二个 future 结合使用，该 future 适用于您正在使用的异步环境。

Maybe there is something wrong with what I said before. What I want to express is that bulk_out returns the number of bytes that have been written, rather than returning the ownership of the buffer. As for the writing speed, it is a little slow because nusb uses winusb under the window platform

martinling, 2024-08-03:
I don't know why you think that returning ownership of the buffer slows anything down.

A `Vec` is just a pointer and a length. Returning a pointer and a length, versus just a length, is not something that's going to have any meaningful effect on performance here.

WinUSB is also not a bottleneck. It's more than capable of completely saturating the outbound bandwidth of a USB bus.

What is the scenario in which you're seeing slow write speeds, and why do you think nusb is the cause?

Are you writing data that's spread across multiple transfers? If so, then to maximise speed you'll need to be using the [Queue API](https://docs.rs/nusb/latest/nusb/transfer/struct.Queue.html) `bulk_out_queue`, rather than calling `bulk_out` for one transfer at a time.

kevinmehall, 2024-08-03:
Additionally, if it didn't return ownership of the buffer, it would have to do more work to drop it.

One source of confusion may be that rustdoc does not make the return type very obvious because it just shows the `TransferRequest::Response` associated type on `TransferFuture` -- if it's not clear, the  `bulk_out` future returns a [`ResponseBuffer`](https://docs.rs/nusb/latest/nusb/transfer/struct.ResponseBuffer.html), which only exposes the `actual_length`  but is basically a `Vec`, and owns the memory like a Vec, and can be trivially converted into a Vec. 

If you think there is a performance issue, a more useful bug report would include things like:

* What is the device you're communicating with, and what is its connection speed (Low, High, Full, Super)?
* What transfer rate are you seeing and how did you measure it?
* What transfer rate are you expecting and why?
* What buffer size are you using?

closed 2024-09-21 by kevinmehall