kevinmehall/nusb #78: Examples on how to write and read from a microcontroller via the SerialBus

djhunter67, 2024-08-26:
# Problem
- Write a char or a string to a microcontroller.
- Read a vector of u8 from a SerialPort

# User Case
I have a Teensy4.1 and I can easily discover the device using __nusb__.  I need to check that I can communicate with the serial device.  I need to send ``'s'`` to be received by code I have written and is running on the microcontroller.  

Then, read the returning value from the microcontroller, which can be up to 512 bytes of information.

# Proposed Solution
## Write
- A ``write`` method.
## Read
- A ``read`` method

kevinmehall, 2024-08-26:
If your device exposes a serial port, consider using your operating system's USB-serial driver via a library like https://docs.rs/serial2/0.2.26/serial2/ or https://docs.rs/serialport5/5.0.2/serialport5/index.html

nusb is more for talking to vendor-class devices that don't have a kernel driver. You could implement a serial port class with nusb, but it'd be easier to use the existing kernel driver.

closed 2024-08-26 by kevinmehall

djhunter67, 2024-08-26:
I understand what you are saying.  To be clear, do you mean the ``nusb`` is not designed to communicate to serial devices?

kevinmehall, 2024-08-26:
nusb is for lower-level access to USB devices at the USB level (interfaces, endpoints, transfers, etc). USB serial (probably the [CDC ACM](https://cscott.net/usb_dev/data/devclass/usbcdc11.pdf) class) is a higher-level protocol on top of USB, and to use that, you want a serial port API, not a USB API.