kevinmehall/nusb #79: Equivalent of rusb interface descriptor extra()?

mlukaszek, 2024-09-06:
Hi,

Firstly, thank you for this project. I love that it is licensed differently to rusb, as statically compiling libusb in is problematic.

I am wondering what is the recommended way of accessing the "extra" bytes from the interface descriptor in nusb?

Similarly to:
https://github.com/a1ien/rusb/blob/master/src/interface_descriptor.rs#L115

Many thanks

mlukaszek, 2024-09-06:
I realize I can call get_descriptor() on a device to receive a raw config descriptor and then search inside the Vec<u8> that's returned until I find the specific data I'm interested in, but wondering if there isn't any easier way.

tuna-f1sh, 2024-09-06:
I also wondered this but found the best way to get the equivalent of "extra" bytes to to filter the `descriptors()`. One must filter as the endpoints are there too. `skip(1)` because first is the interface itself. Works on config too though you'd filter on 0x02:

```
let interface_extra = interface_alt
    .descriptors()
    .skip(1)
    // only want device and interface descriptors - nusb has everything trailing including endpoint
    .filter(|d| {
        (d.descriptor_type() & 0x0F) == 0x04 || (d.descriptor_type() & 0x0F) == 0x01
    })
```

Don't need open the device to use `get_descriptor()` with this.

kevinmehall, 2024-09-06:
Yes, [Device::active_configuration](https://docs.rs/nusb/latest/nusb/struct.Device.html#method.active_configuration) or [Interface::descriptors](https://docs.rs/nusb/latest/nusb/struct.Interface.html#method.descriptors) provides parsed descriptor access, and then the [descriptors](https://docs.rs/nusb/latest/nusb/descriptors/struct.InterfaceAltSetting.html#method.descriptors) method of most of the parsed types allows iterating all descriptors under it, including unknown extra descriptors.

mlukaszek, 2024-09-06:
This is great. I will try it next week and report back here - thanks!

mlukaszek, 2024-09-09:
I can confirm it worked without any problems. Using flat_map and then collecting a Vec<u8> is the main part of the trick I was missing. Thanks again!