kevinmehall/nusb #83: WebUSB backend

kevinmehall, 2024-09-17:
https://github.com/G2-Games/cross-usb wraps `nusb` and WebUSB with a subset of the API, but  WebUSB could be an `nusb` backend in-tree instead of requiring a wrapper library. 

This is probably not something I'll get to soon, but the API implications are worth considering, mainly that *all* WebUSB operations are async, while nusb currently has many blocking methods where the underlying OS operations are blocking.

For non-Wasm, it looks like `cross_usb` just calls the `nusb` blocking methods from an `async fn`, blocking the async runtime. Probably fine for a lot of use cases, but not ideal if your application is primarily async. Doing that right would require depending on [blocking](https://crates.io/crates/blocking) or [`tokio::spawn_blocking`](https://docs.rs/tokio/latest/tokio/task/fn.spawn_blocking.html). So far nusb has managed to avoid depending on an async executor.

(cc @G2-Games)

G2-Games, 2024-09-17:
I thought about looking into including WASM support in `nusb` originally, but I thought it would be best to do in a separate crate for the moment as my main use case was just a single library I'm working on (`minidisc-rs`). However, if you're interested in it that's very cool! I'm happy to help or modify my code to be better/work properly, but I haven't looked into the internals of `nusb`.

As you say, WASM cannot block, which actually led to the refactoring of the entirety of `minidisc-rs` to async due to how `cross_usb` works.

Thank you for your awesome work on `nusb`!

TroyNeubauer, 2024-10-08:
Going to hold off on https://github.com/G2-Games/cross-usb/issues/2 if wasm support is coming to `nusb` in the near future