kevinmehall/nusb #84: Publish new version?

TroyNeubauer, 2024-10-08:
When you have time, could you publish a new version?

I didn't realize you cant publish to crates.io with git dependencies, and I'm looking to publish https://github.com/MerchGuardian/seify-hackrfone soon.

kevinmehall, 2024-10-09:
I assume you're most interested in your Android change? I have some small breaking changes on `main` as I was planning for the next version to be v0.2. The biggest change in v0.2 is planned to be a rework of the endpoint API, which I haven't had time to finish.

But I can backport https://github.com/kevinmehall/nusb/pull/80 to a v0.1.11 release soon.

TroyNeubauer, 2024-10-09:
Yes, android in particular. Backport would be great, but dont break your back over it

kevinmehall, 2024-11-03:
Published [v0.1.11](https://github.com/kevinmehall/nusb/releases/tag/v0.1.11) with `Device::from_fd` and other changes that could be rebased easily. Other changes on `main` that were too entangled with the Windows enumeration rewrite will have to wait for v0.2.

closed 2024-11-03 by kevinmehall

tuna-f1sh, 2024-11-06:
@kevinmehall happy to help with anything I can for v0.2. Is the Windows enumeration re-write something you're working on locally? Have to say, Windows is my least accessible platform for develop/test however!