kevinmehall/nusb #89: next_complete waiting forever although the completion callback have been emitted

hrouault, 2024-11-08:
Hi,

I am trying to use nusb with the cameleon library for usb3 cameras. Sometimes, the program hangs because the next_complete method never returns.

Here is how I read the packets:

```rust
    async fn read_leader(&mut self) -> StreamResult<()> {
        let leader_buf = self.queue.next_complete().await.into_result()?;
        self.leader_buf = Some(leader_buf);
        Ok(())
    }
```
I checked that no error is returned and that the program hangs on the await.

Here is a log where the packets are well received for a transfer and then it hangs for the next one.

```log
2024-11-08T13:04:40.279223Z DEBUG nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600000d2a520, status=0, len=52
2024-11-08T13:04:40.279280Z DEBUG nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600000d24520, status=0, len=65536
2024-11-08T13:04:40.279371Z DEBUG cameleon::u3v::stream_handle: Received the leader
2024-11-08T13:04:40.279400Z DEBUG cameleon::u3v::stream_handle: Waiting for completion: 0, 7 pending transfers
2024-11-08T13:04:40.279451Z DEBUG cameleon::u3v::stream_handle: Waiting for completion: 65536, 6 pending transfers
2024-11-08T13:04:40.279584Z DEBUG nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600000d24ba0, status=0, len=65536
2024-11-08T13:04:40.279675Z DEBUG cameleon::u3v::stream_handle: Waiting for completion: 131072, 5 pending transfers
2024-11-08T13:04:40.279923Z DEBUG nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600000d24a80, status=0, len=65536
2024-11-08T13:04:40.279991Z DEBUG cameleon::u3v::stream_handle: Waiting for completion: 196608, 4 pending transfers
2024-11-08T13:04:40.280272Z DEBUG nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600000d243c0, status=0, len=65536
2024-11-08T13:04:40.280339Z DEBUG cameleon::u3v::stream_handle: Waiting for completion: 262144, 3 pending transfers
2024-11-08T13:04:40.280633Z DEBUG nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600000d246a0, status=0, len=65536
2024-11-08T13:04:40.280700Z DEBUG cameleon::u3v::stream_handle: Waiting for completion: 327680, 2 pending transfers
2024-11-08T13:04:40.280951Z DEBUG nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600000d24460, status=0, len=61120
2024-11-08T13:04:40.280982Z DEBUG nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600000d248a0, status=0, len=32
2024-11-08T13:04:40.281011Z DEBUG cameleon::u3v::stream_handle: Received the main payload
2024-11-08T13:04:40.281027Z DEBUG cameleon::u3v::stream_handle: Waiting for completion trailer, 1 pending transfers
2024-11-08T13:04:40.281038Z DEBUG cameleon::u3v::stream_handle: Received all the packets
2024-11-08T13:04:40.281227Z DEBUG nusb::platform::macos_iokit::transfer: Submitted IN transfer 0x600000d248a0 on endpoint 81
2024-11-08T13:04:40.281303Z DEBUG nusb::platform::macos_iokit::transfer: Submitted IN transfer 0x600000d24460 on endpoint 81
2024-11-08T13:04:40.281352Z DEBUG nusb::platform::macos_iokit::transfer: Submitted IN transfer 0x600000d246a0 on endpoint 81
2024-11-08T13:04:40.281397Z DEBUG nusb::platform::macos_iokit::transfer: Submitted IN transfer 0x600000d243c0 on endpoint 81
2024-11-08T13:04:40.281457Z DEBUG nusb::platform::macos_iokit::transfer: Submitted IN transfer 0x600000d24a80 on endpoint 81
2024-11-08T13:04:40.281515Z DEBUG nusb::platform::macos_iokit::transfer: Submitted IN transfer 0x600000d24ba0 on endpoint 81
2024-11-08T13:04:40.281565Z DEBUG nusb::platform::macos_iokit::transfer: Submitted IN transfer 0x600000d24520 on endpoint 81
2024-11-08T13:04:40.281619Z DEBUG nusb::platform::macos_iokit::transfer: Submitted IN transfer 0x600000d24880 on endpoint 81
2024-11-08T13:04:40.281632Z DEBUG cameleon::u3v::stream_handle: trailer size: 56
2024-11-08T13:04:40.281642Z DEBUG cameleon::u3v::stream_handle: Submitted all the packets
2024-11-08T13:04:40.294321Z DEBUG nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600000d248a0, status=0, len=52
2024-11-08T13:04:40.294422Z DEBUG cameleon::u3v::stream_handle: Received the leader
2024-11-08T13:04:40.294421Z DEBUG nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600000d24460, status=0, len=65536
2024-11-08T13:04:40.294455Z DEBUG cameleon::u3v::stream_handle: Waiting for completion: 0, 7 pending transfers
2024-11-08T13:04:40.294765Z DEBUG nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600000d246a0, status=0, len=65536
2024-11-08T13:04:40.295098Z DEBUG nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600000d243c0, status=0, len=65536
2024-11-08T13:04:40.295461Z DEBUG nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600000d24a80, status=0, len=65536
2024-11-08T13:04:40.295786Z DEBUG nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600000d24ba0, status=0, len=65536
2024-11-08T13:04:40.296131Z DEBUG nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600000d24520, status=0, len=61120
2024-11-08T13:04:40.296165Z DEBUG nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600000d24880, status=0, len=32
```

Is there something I should pay attention to?

kevinmehall, 2024-11-08:
Hmm, if you are able to dive in and troubleshoot this, here's a tour of the code involved.

You're seeing this log message:
https://github.com/kevinmehall/nusb/blob/c4395fe293a7d1e52f6657421b4628fe94475be8/src/platform/macos_iokit/transfer.rs#L21-L36

When that calls `notify_completion`, it should find the transfer in STATE_PENDING, update it to STATE_COMPLETED and wake the waker:
https://github.com/kevinmehall/nusb/blob/c4395fe293a7d1e52f6657421b4628fe94475be8/src/transfer/internal.rs#L176-L196

The waker should have been previously registered here, which means that async task should wake up and poll again, and find the transfer in STATE_COMPLETED
https://github.com/kevinmehall/nusb/blob/c4395fe293a7d1e52f6657421b4628fe94475be8/src/transfer/internal.rs#L128-L144

You could start by adding additional logging to these (or using a debugger)  to see which of those steps is not happening.

hrouault, 2024-11-15:
Hi,
Thanks for pointing me to the relevant functions. I didn't have time to look at it before but here are the results:

I've added logging in the function `notify_completion` and `poll_completion_generic`:

```
2024-11-15T09:45:50.086299Z DEBUG cameleon::u3v::stream_handle: Waiting for completion: 0, 7 pending transfers
2024-11-15T09:45:50.086324Z  INFO nusb::transfer::internal: state is completed
2024-11-15T09:45:50.086390Z DEBUG cameleon::u3v::stream_handle: Waiting for completion: 65536, 6 pending transfers
2024-11-15T09:45:50.086445Z  INFO nusb::transfer::internal: state is completed
2024-11-15T09:45:50.086501Z  INFO nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600001ad3e60, status=0, len=65536
2024-11-15T09:45:50.086517Z DEBUG cameleon::u3v::stream_handle: Waiting for completion: 131072, 5 pending transfers
2024-11-15T09:45:50.086551Z  INFO nusb::transfer::internal: state is pending
2024-11-15T09:45:50.086585Z  INFO nusb::transfer::internal: The transfer is pending
2024-11-15T09:45:50.086788Z  INFO nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600001ae3b60, status=0, len=65536
2024-11-15T09:45:50.086809Z  INFO nusb::transfer::internal: The transfer is pending
2024-11-15T09:45:50.087276Z  INFO nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600001ac0ea0, status=0, len=65536
2024-11-15T09:45:50.087296Z  INFO nusb::transfer::internal: The transfer is pending
2024-11-15T09:45:50.087492Z  INFO nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600001adf4e0, status=0, len=61120
2024-11-15T09:45:50.087515Z  INFO nusb::transfer::internal: The transfer is pending
2024-11-15T09:45:50.087535Z  INFO nusb::platform::macos_iokit::transfer: Completion callback for transfer 0x600001a30320, status=0, len=32
2024-11-15T09:45:50.087547Z  INFO nusb::transfer::internal: The transfer is pending
```

The `notify_completion` seems to function as expected (the state was pending and the waker is called). However, the `poll_completion_generic` function is never called.

I would appreciate your further guidance. thanks

kevinmehall, 2024-11-15:
You should expect to see `poll_completion_generic` called at least twice for each `next_complete().await` -- once when the future is first polled, which will likely see `STATE_PENDING`.  This will register the task's waker, so the task gets woken when the transfer is ready. When woken, the task will poll again, and see `STATE_COMPLETED` and return the data.

If it is not being called at all, I have to suspect something about how you are polling the returned future.

The path from `next_complete` to `poll_completion_generic` is here:

https://github.com/kevinmehall/nusb/blob/c4395fe293a7d1e52f6657421b4628fe94475be8/src/transfer/queue.rs#L158-L192

The only thing I could see going wrong there is if `front_mut()` is the wrong transfer, such as if they somehow complete out of order, but even then, you'd still see it poll the first time before getting stuck.

What executor are you using / how are you polling the future or the future that `.await`s it?

hrouault, 2024-11-15:
This was not clear in my first message, but this problem arises rarely: it hangs after thousand of successful transfers.

From what I see in successful transfers, the `poll_completion_generic` function is not always called twice for each next_complete().await: sometimes it is called only once when the transfer is already completed. Is it normal behavior? I thought this was expected, and an advantage of the queue system?

From what I understand, when it hangs, the `poll_completion_generic` function is not called after the `notify_completion`.

How would the transfer complete out of order? How could I check that?

I am used tasks from the iced gui, but it uses tokio under the hood. A thread is dedicated to the queue await s. I poll the future with functions similar to the one in the first code snippet of the first message. I have one such function for each stage of the transfer (3 functions in total, but all very similar).

kevinmehall, 2024-11-16:
I found a race condition that could lead to lost wakeups. Can you try https://github.com/kevinmehall/nusb/pull/92?

hrouault, 2024-11-19:
I cherry-picked the last commit from that branch and it seems to solve the issue (I waited for about 10 min when it was hanging in the first few minutes usually).
Thanks a lot for your reactivity!

kevinmehall, 2024-11-21:
Fix released in [v0.1.12](https://github.com/kevinmehall/nusb/releases/tag/v0.1.12)

PR #92 (2024-11-20): Fix race condition that could lose a transfer completion wakup
If `poll_completion_generic` runs between `notify_completion`'s `waker.take()` and `state.swap(STATE_COMPLETED, ...)`, it could see the transfer as pending and register its waker, but that waker would never be called.

This adds an Arc because once the transfer is set to `STATE_COMPLETED`, the other thread might free it, so we need to ensure the AtomicWaker stays alive while we use it. That's what the `take()` was intended to work around, but reading the waker before writing the status introduces the race.

Fixes #89
closed 2024-11-21 by kevinmehall