kevinmehall/nusb #93: `Queue::submit`, what is the limit on number of transfers?

korken89, 2024-11-22:
Hi!

The documentation of `Queue::submit` is not clear on its limitations, and if one ships e.g. 1M transfers it silently fails.
What are the limitations really? I can fix a PR documenting it.

Thanks for the insight!

kevinmehall, 2024-11-22:
On Linux there is a global limit `/sys/module/usbcore/parameters/usbfs_memory_mb` (default 16mb) for the total size of all in-flight transfers across all devices and all processes.

I do not know of documented limits for Windows and macOS, nor have I tested pushing the limits. If you find documentation or run tests, please let me know.

Windows does expose a API to get a [MAXIMUM_TRANSFER_SIZE](https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/winusb-functions-for-pipe-policy-modification) per transfer that we should probably wrap and expose. It is normally 2MB.

korken89, 2024-11-29:
I dug for a bit but quickly realized I'm in over my head and can unfortunately not spend more time here.
Thank you for the insight!

closed 2024-11-29 by korken89