kevinmehall/nusb #98: Windows and the `interfaces()` method

jamesmunns, 2024-12-10:
Hey there, in https://docs.rs/nusb/latest/nusb/struct.DeviceInfo.html#method.interfaces, it mentions that unless the device is using the generic driver, it will not return any interfaces.

For what I'm doing, I'm selecting the WinUSB driver, so on windows I don't see any interfaces.

Do you know how to get around this on Windows? Would opening a connection to the device and using [Device::active_configuration](https://docs.rs/nusb/latest/nusb/struct.Device.html#method.active_configuration) is possible?

kevinmehall, 2024-12-10:
Yes, the `Device` methods parse the descriptors and will see all interfaces. On `main` / upcoming 0.2, it falls back to using cached descriptors for `DeviceInfo` too, working around that caveat (won't have interface string descriptors cached, though).

There are two ways to set up WinUSB. One binds the whole device to WinUSB, and Windows sees only one device object in its driver model. The other binds usbccgp (Composite Group Parent) to the device, which then creates Windows device objects for each interface. Those interfaces can be bound to WinUSB, or to another driver like HID or Serial. Generally a composite device will use the latter option, but if all interfaces would be accessed with WinUSB, the former is also possible.

In the postcard-rpc case, you probably want to support both. If you're given a specific `DeviceInfo` or find one by VID/PID/etc, then you're fine: open the device and use `active_configuration` to find the postcard-rpc interface and its endpoints. `DeviceInfo::interfaces()` is meant for finding a device based on its interfaces (e.g. searching for a particular interface class), where you don't necessarily want to open each device and probe it.

jamesmunns, 2024-12-10:
Yeah, we recently added the ability to scan for interfaces, to allow composite devices (postcard-rpc AND midi, for example), but I got tripped up because on NON composite devices, we can't filter on the interfaces at all.

At least on Windows, or maybe everywhere, I should probably make it a two-stage process: do a coarse filter on the DeviceInfo (product name, vid/pid, etc), then attempt to connect and find a matching interface.

Thanks for the info! I definitely need to rework the "find a postcard-rpc device" logic, as in the other issue you got tagged in, I need to be a bit more robust in finding things.

jamesmunns, 2024-12-10:
Closing this, as it's more of a Q&A than issue as everything is working correctly. Thanks for the insight!

closed 2024-12-10 by jamesmunns