kevinmehall/nusb #101: GetDeviceInfo from file descriptor

ratzrattillo, 2024-12-30:
I could not find a way to get a DeviceInfo struct from a file descriptor (like the one that can be used in Device::from_fd()).
That would be really helpful in situations where only the FileDescriptor is given, like on Android, right?

kevinmehall, 2024-12-30:
Android doesn't allow access to sysfs, which we use to populate DeviceInfo on Linux. You can get some of it from the device descriptor that can be read from the usbfs FD, but to get more, you need to go through Java APIs as prototyped here:

* https://github.com/kevinmehall/nusb/issues/86#issuecomment-2530424108
* https://docs.rs/android-usbser/0.2.1/src/android_usbser/usb_info.rs.html

ratzrattillo, 2025-01-01:
Thanks for the clarifications!

closed 2025-01-01 by ratzrattillo