kevinmehall/nusb #106: Listing devices not working on linux (PopOS!)

andrewdavidmackenzie, 2025-01-10:
I use nusb in my app to discover and interact with my devices (Pi Picos) via USB.

For example, I use this code to list devices and filter by my own vendor id:
https://github.com/andrewdavidmackenzie/pigg/blob/master/src/usb.rs#L236

This has worked perfectly on my main development machine/OS (macos 15) from the start and is easy to use.

I just started testing on Linux (PopOS) and the same code doesn't list/detect the devices.

Any help in understanding this issue on linux would be appreciated (Windows will come later...)

thanks

andrewdavidmackenzie, 2025-01-10:
Using `lsusb` on linux shows the device correctly with the expected vendor and product IDs.

andrewdavidmackenzie, 2025-01-10:
It seems the error comes later (not list devices).
I am getting a permission denied error when I try to open the device to query details.
"Permission denied (os error 13)"

kevinmehall, 2025-01-10:
Does your user have write permissions to access the device? `ls -l /dev/bus/usb/*/*` to check the owner and permissions. The numbers in the path correspond to the bus number and device address seen by nusb, also shown in `lsusb`.

See the [Linux section of the docs](https://docs.rs/nusb/latest/nusb/#linux) for how to create a udev rule to configure device permissions.

andrewdavidmackenzie, 2025-01-10:
```
ls -l /dev/bus/usb/*/*
crw-rw-r-- 1 root root 189,   0 Jan 10 09:14 /dev/bus/usb/001/001
crw-rw-r-- 1 root root 189,   2 Jan 10 09:14 /dev/bus/usb/001/003
crw-rw-r-- 1 root root 189,   3 Jan 10 10:02 /dev/bus/usb/001/004
crw-rw-r-- 1 root root 189,   4 Jan 10 09:14 /dev/bus/usb/001/005
crw-rw-r-- 1 root root 189,   8 Jan 10 10:52 /dev/bus/usb/001/009
crw-rw-r-- 1 root root 189, 128 Jan 10 09:14 /dev/bus/usb/002/001
crw-rw-r-- 1 root root 189, 256 Jan 10 09:14 /dev/bus/usb/003/001
crw-rw-r-- 1 root root 189, 384 Jan 10 09:14 /dev/bus/usb/004/001
```

They all belong to root.

Is this normal? For users of my app will I have to ask them to drop down to the terminal and as root get them to add permissions to those device files in order to acces smy device over USB?

andrewdavidmackenzie, 2025-01-10:
IIUC, I should instruct users to add

```
SUBSYSTEM=="usb", ATTRS{idVendor}=="ZZZZ", ATTRS{idProduct}=="ZZZZ", MODE="0660", GROUP="plugdev"
```
(with my vendor and product ids) 

to: /etc/udev/rules.d/70-plugdev-usb.rules?

Is there a reliable way to detect this case (match on the error returned to being a permission error?) and advise users to do that?

andrewdavidmackenzie, 2025-01-10:
So now I am part of plugdev
```
groups
andrew adm sudo plugdev lpadmin wireshark
```

but still lack access, probably due to those devices being in the root group?

kevinmehall, 2025-01-10:
All your devices are owned by root, as is the default on most Linux distros.

If your app is installed systemwide with a package manager, your package can install a udev rules file to change the default permissions for your device. Otherwise, yes, you have to get users to install it manually.

In your udev rule, the VID and PID should be in hex, not decimal, and you should use a filename like `70-pigg.rules`. All files in `/etc/udev/rules.d/` are processed in sorted order. Note that the `plugdev` group is a Debian/Ubuntu thing, and may not exist on other distros. Some documentation recommends specifying `TAG+="uaccess"` instead of specifying a group to allow access to the logged-in user, but this only works for the graphical session and not over SSH.

Yes, the permission denied error is the way to check for this.

andrewdavidmackenzie, 2025-01-10:
OK, thanks for your help - I will close this issue now.

closed 2025-01-10 by andrewdavidmackenzie