kevinmehall/nusb #107: The non-main thread is unable to retrieve the interfaces properly

lethargy123, 2025-01-11:
I used tokio::task::spawn to create a new thread, and then inside this new thread, I called the nusb::list_devices function. However, the DeviceInfo's interfaces obtained in this thread are empty (there's no issue when I directly use nusb::list_devices in the main thread).

lethargy123, 2025-01-11:
The computer system is window11

kevinmehall, 2025-01-12:
I'm not seeing anything break when wrapping the [`list` example](https://github.com/kevinmehall/nusb/blob/main/examples/list.rs) in `std::thread::spawn(|| { }).join().unwrap();`. Could you share a complete minimal example that exhibits this behavior? And also a debug log, e.g. by using `env_logger` and setting `RUST_LOG=nusb=debug`.

lethargy123, 2025-01-14:
I am unable to reproduce this bug, it might be unstable or caused by other factors (when I ran the code on that day, the printed interfaces were occasionally empty). I will temporarily close this issue for now.

closed 2025-01-14 by lethargy123