kevinmehall/nusb #119: Acquire mount point of storage device

terrarier2111, 2025-03-03:
Is there a way to get the mount point of a device (specifically if its a storage device)?

kevinmehall, 2025-03-07:
nusb is primarily for writing user-space drivers for custom devices that don't have a kernel driver. While you can list mass storage devices at the USB level, it doesn't have anything for associating them with block devices or mount points because those involve other kernel drivers on top of USB. I think it's out of scope for nusb, though I think there's a place for a cross platform library for listing devices and exposing bus topology relations, which could re-use a lot of nusb's cfgmgr32 / sysfs / iokit enumeration code, and interoperate with nusb.

tuna-f1sh, 2025-03-08:
Would be a potential feature for cyme I think: https://github.com/tuna-f1sh/cyme/issues/60

closed 2025-03-08 by kevinmehall