kevinmehall/nusb #120: Seeking advice on control_in: interface or device

andrewdavidmackenzie, 2025-03-06:
I have an app that "discovers" attached devices (pi picos with my app) by USB.

I was using this code: https://github.com/andrewdavidmackenzie/pigg/blob/master/pignet/src/usb_host.rs#L235

to detect the matching devices, get their serial numbers, and then query each one for a description struct of mine, using interface::control_in

This was working fine, until I attached two devices via USB. Detection of one was failing.

I first thought it might be the device::claim_interface() in the loop. But debugging, that is not failing, it's the later control_in call that fails.

I did a hack locally to see if using device::control_in changed anything, and it seems to solve the problem (although then I'm not using the interface::set_alt_setting(1) call.

Basically, I'm looking for advice/direction on when to use device and when to use interface be able to use control_in to get information from the device.

The docs (https://docs.rs/nusb/latest/nusb/struct.Device.html#method.control_in) didn't really clarify that for me.

thanks!

andrewdavidmackenzie, 2025-03-06:
I just saw 

```Not supported on Windows. You must [claim an interface](https://docs.rs/nusb/latest/nusb/struct.Device.html#method.claim_interface) and use the interface handle to submit transfers.```

So, maybe that's one part of it - I do support Windows so that might be an issue to use device::control_in always.

kevinmehall, 2025-03-06:
On Linux and macOS, there is no difference -- they call the same implementation. The Device methods exist so you can use them without claiming an interface on platforms that support it, while the Interface methods exist for applications that only hold an Interface, and because WinUSB requires a claimed interface.

So that doesn't explain why one works and the other doesn't. When it fails, what is the error? Can you post a debug log? (set `RUST_LOG=nusb=debug` with env-logger).

andrewdavidmackenzie, 2025-03-07:
That's annoying. Now I can't reproduce it. I've connected an additional (different) device by USB, but not remove any of the previous ones - and now it seems to be working. I'll leave this issue open for a short time until I can reproduce it, otherwise I will close (if that is OK).

andrewdavidmackenzie, 2025-03-12:
Saw this again today (will attempt to reproduce reliably). Message was "Could not be opened for exclusive access".

I assume this is my issue, using claim_interface() within a loop over connected devices?
https://github.com/andrewdavidmackenzie/pigg/blob/master/pignet/src/usb_host.rs#L253

kevinmehall, 2025-03-16:
That error would be expected if the interface of that device is already claimed, whether in another process, the same process, or by a kernel driver.

closed 2025-05-03 by kevinmehall