kevinmehall/nusb #126: Request to add custom USB device functionality

lethargy123, 2025-04-11:
I want to use a custom USB driver under Windows instead of WinUSB or other standard drivers. I noticed there are similar implementations in the code. Is it possible to create a wrapper function that accepts parameters like a handle to the custom USB device or other identifiers?

kevinmehall, 2025-04-12:
Could you give more detail on what you are trying to do? What does your custom driver do?

Most of nusb on Windows uses WinUSB APIs and won't work without WinUSB. The only part that does not require WinUSB is `list_devices`. That can already give you the [instance ID](https://docs.rs/nusb/latest/x86_64-pc-windows-msvc/nusb/struct.DeviceInfo.html#method.instance_id) which you can use with cfgmgr32 or SetupAPI functions to access another driver on the device.

lethargy123, 2025-04-25:
> 您能否详细介绍您正在努力做什么？您的自定义驱动程序有什么作用？
> 
> Windows 上的大多数 nusb 都使用 WinUSB API，没有 WinUSB 就无法工作。唯一不需要 WinUSB 的部分是 .这已经可以为您提供[实例 ID](https://docs.rs/nusb/latest/x86_64-pc-windows-msvc/nusb/struct.DeviceInfo.html#method.instance_id)，您可以将其与 cfgmgr32 或 SetupAPI 函数一起使用，以访问设备上的另一个驱动程序。`list_devices`

In fact, the custom USB device I mentioned does not use WinUSB and has a very streamlined implementation. I found a C++ example online that first uses WinAPI to obtain a device handle, then directly employs bulk transfers for read/write operations. Of course, different devices require distinct approaches. I believe rusb could expose more methods to help USB developers better customize USB devices. For instance, my USB device might use either the WinUSB driver or another driver. In such cases, I could use nusb with alternative drivers instead of invoking Zadig to modify the device driver or developing a separate driver adaptation.

[image]

kevinmehall, 2025-04-25:
There is not very much from nusb that would be useful for this. Even how to find the device path to open the handle might be different if you are not using WinUSB. If you want to use a different driver, you're going to need to write different code to use it. Look at the `windows-sys` crate to use Windows APIs from Rust. To support both, I'd suggest an abstraction layer on top of nusb and your custom driver interface.

closed 2025-04-25 by kevinmehall