kevinmehall/nusb #129: macos : EXC_BAD_ACCESS crash inside `cancel`

xobs, 2025-05-10:
I'm trying to track this down, but I'm seeing a crash inside `nusb`. The stack trace looks like this:

```
(lldb) bt
* thread #12, name = 'tokio-runtime-worker', stop reason = EXC_BAD_ACCESS (code=1, address=0x823)
    frame #0: 0x0000000101accba0 probe-rs`_$LT$nusb..platform..macos_iokit..transfer..TransferData$u20$as$u20$nusb..transfer..internal..PlatformTransfer$GT$::cancel::h865fee429fdae0e5(self=0x0000600002690f48) at iokit.rs:74:20
  * frame #1: 0x0000000101ab5efc probe-rs`nusb::transfer::internal::TransferHandle$LT$P$GT$::cancel::h1fa1f20f26a50019(self=0x0000000171451258) at internal.rs:128:9
    frame #2: 0x0000000101ab5fc0 probe-rs`_$LT$nusb..transfer..internal..TransferHandle$LT$P$GT$$u20$as$u20$core..ops..drop..Drop$GT$::drop::hfb13c8f76062b439(self=0x0000000171451258) at internal.rs:167:17
    frame #3: 0x0000000101aa568c probe-rs`core::ptr::drop_in_place$LT$nusb..transfer..internal..TransferHandle$LT$nusb..platform..macos_iokit..transfer..TransferData$GT$$GT$::h2784156c86d9a052((null)=0x0000000171451258) at mod.rs:523:1
    frame #4: 0x00000001014599b4 probe-rs`core::mem::drop::ha3a88dcfddc2fccc(_x=TransferHandle<nusb::platform::macos_iokit::transfer::TransferData> @ 0x0000000171451258) at mod.rs:940:24
    frame #5: 0x00000001013f3930 probe-rs`core::ops::function::FnMut::call_mut::h3e60212b6b981ed7((null)=0x00000001714512e7, (null)=(nusb::transfer::internal::TransferHandle<nusb::platform::macos_iokit::transfer::TransferData>) @ 0x0000000171451270) at function.rs:166:5
    frame #6: 0x0000000101795ac8 probe-rs`core::iter::traits::iterator::Iterator::for_each::call::_$u7b$$u7b$closure$u7d$$u7d$::hb94309ada1bf0abd((null)=<unavailable>, item=TransferHandle<nusb::platform::macos_iokit::transfer::TransferData> @ 0x00000001714512a8) at iterator.rs:797:29
    frame #7: 0x0000000101803e90 probe-rs`core::iter::traits::double_ended::DoubleEndedIterator::rfold::h071cad27c5f62770(self=Drain<nusb::transfer::internal::TransferHandle<nusb::platform::macos_iokit::transfer::TransferData>, alloc::alloc::Global> @ 0x0000000171451330, init=<unavailable>, f={closure_env#0}<nusb::transfer::internal::TransferHandle<nusb::platform::macos_iokit::transfer::TransferData>, fn(nusb::transfer::internal::TransferHandle<nusb::platform::macos_iokit::transfer::TransferData>)> @ 0x00000001714512e7) at double_ended.rs:308:21
    frame #8: 0x00000001016dafe4 probe-rs`_$LT$core..iter..adapters..rev..Rev$LT$I$GT$$u20$as$u20$core..iter..traits..iterator..Iterator$GT$::fold::hc663f243ac98599e(self=<unavailable>, init=<unavailable>, f={closure_env#0}<nusb::transfer::internal::TransferHandle<nusb::platform::macos_iokit::transfer::TransferData>, fn(nusb::transfer::internal::TransferHandle<nusb::platform::macos_iokit::transfer::TransferData>)> @ 0x000000017145135f) at rev.rs:64:9
    frame #9: 0x00000001016da564 probe-rs`core::iter::traits::iterator::Iterator::for_each::h9d28673d2dc0c5ee(self=<unavailable>, f=0x0000017145140000) at iterator.rs:800:9
    frame #10: 0x000000010140fb20 probe-rs`_$LT$nusb..transfer..queue..Queue$LT$R$GT$$u20$as$u20$core..ops..drop..Drop$GT$::drop::hd950589ba86e883f(self=0x0000000171452578) at queue.rs:233:9
    frame #11: 0x00000001014087d4 probe-rs`core::ptr::drop_in_place$LT$nusb..transfer..queue..Queue$LT$nusb..transfer..buffer..RequestBuffer$GT$$GT$::hadde3db2f62d0523((null)=0x0000000171452578) at mod.rs:523:1
    frame #12: 0x00000001015a9aa0 probe-rs`_$LT$nusb..device..Interface$u20$as$u20$probe_rs..probe..usb_util..InterfaceExt$GT$::read_bulk::_$u7b$$u7b$closure$u7d$$u7d$::h7e0d041d492011d4((null)=0x0000000171452618) at usb_util.rs:53:13
    frame #13: 0x00000001013eed10 probe-rs`_$LT$futures_lite..future..Or$LT$F1$C$F2$GT$$u20$as$u20$core..future..future..Future$GT$::poll::hc8256e4285276086(self=Pin<&mut futures_lite::future::Or<probe_rs::probe::usb_util::{impl#0}::read_bulk::{async_block_env#0}, probe_rs::probe::usb_util::{impl#0}::read_bulk::{async_block_env#1}>> @ 0x0000000171452000, cx=0x0000000171452618) at future.rs:454:33
    frame #14: 0x000000010183d194 probe-rs`async_io::driver::block_on::_$u7b$$u7b$closure$u7d$$u7d$::h256d6f0ddd02f6d8(cache=0x0000000123834860) at driver.rs:199:37
    frame #15: 0x00000001014efe54 probe-rs`std::thread::local::LocalKey$LT$T$GT$::try_with::h023a3df241f73d62(self=0x00000001033fa310, f={closure_env#1}<core::result::Result<usize, std::io::error::Error>, futures_lite::future::Or<probe_rs::probe::usb_util::{impl#0}::read_bulk::{async_block_env#0}, probe_rs::probe::usb_util::{impl#0}::read_bulk::{async_block_env#1}>> @ 0x0000000171453a18) at local.rs:310:12
    frame #16: 0x00000001014efb98 probe-rs`std::thread::local::LocalKey$LT$T$GT$::with::h551ac6f465910a6f(self=0x00000001033fa310, f=<unavailable>) at local.rs:274:15
    frame #17: 0x000000010183ca40 probe-rs`async_io::driver::block_on::hbbd0c9a9db27281d(future=Or<probe_rs::probe::usb_util::{impl#0}::read_bulk::{async_block_env#0}, probe_rs::probe::usb_util::{impl#0}::read_bulk::{async_block_env#1}> @ 0x0000000171453cd0) at driver.rs:175:5
    frame #18: 0x0000000101658cb0 probe-rs`_$LT$nusb..device..Interface$u20$as$u20$probe_rs..probe..usb_util..InterfaceExt$GT$::read_bulk::h08a96f549da3cf25(self=0x0000000122f2a8a0, endpoint='\x83', buf=size=4096, timeout=Duration @ 0x0000000171453c48) at usb_util.rs:64:9
    frame #19: 0x00000001016b6da4 probe-rs`probe_rs::probe::xds110::usb_interface::Xds110UsbDevice::send_command::hfa1d7224771daa9c(self=0x0000000122f2a8a0, cmd=JtagScan @ 0x0000000171457100) at usb_interface.rs:127:26
    frame #20: 0x00000001018005e8 probe-rs`_$LT$probe_rs..probe..xds110..Xds110$u20$as$u20$probe_rs..probe..JtagAccess$GT$::write_register::h8d0d03f3d345f03e(self=0x0000000122f2a800, address=10, data=size=8, len=35) at mod.rs:1035:22
    frame #21: 0x0000000101878be8 probe-rs`probe_rs::probe::JtagAccess::write_register_batch::hb6d7b9c50072aa55(self=0x0000000122f2a800, writes=0x0000000171458320) at probe.rs:1366:27
[...stack continues...]
```

It seems to happen when transmitting a lot of data -- usually more than 4096 bytes. I don't think this is wrapped in any sort of unsafe, but I'll see if I can't make it more reproducible.

xobs, 2025-05-10:
I'm using this routine, adapted from `probe-rs`, to read and write the device:

```
use async_io::{Timer, block_on};
use futures_lite::FutureExt;
use nusb::{Interface, transfer::RequestBuffer};
use std::{io, time::Duration};

pub trait InterfaceExt {
    fn read_bulk(&self, endpoint: u8, buf: &mut [u8], timeout: Duration) -> io::Result<usize>;
    fn write_bulk(&self, endpoint: u8, buf: &[u8], timeout: Duration) -> io::Result<usize>;
}

impl InterfaceExt for Interface {
    fn write_bulk(&self, endpoint: u8, buf: &[u8], timeout: Duration) -> io::Result<usize> {
        let fut = async {
            tracing::trace!("Writing data: {:x?}", buf);
            let comp = self.bulk_out(endpoint, buf.to_vec()).await;
            comp.status.map_err(io::Error::other)?;

            let n = comp.data.actual_length();
            Ok(n)
        };

        block_on(fut.or(async {
            Timer::after(timeout).await;
            Err(std::io::ErrorKind::TimedOut.into())
        }))
    }

    fn read_bulk(&self, endpoint: u8, buf: &mut [u8], timeout: Duration) -> io::Result<usize> {
        let fut = async {
            if buf.len() > 64 {
                let mut queue = self.bulk_in_queue(endpoint);
                let n_transfers = buf.len() / 64;
                let transfer_size = buf.len();
                while queue.pending() < n_transfers {
                    queue.submit(RequestBuffer::new(transfer_size));
                }

                let mut offset = 0;
                let mut finished = false;
                while offset < transfer_size && !finished {
                    let comp = queue.next_complete().await;
                    comp.status.map_err(io::Error::other)?;

                    let n = comp.data.len();
                    buf[offset..offset + n].copy_from_slice(&comp.data);
                    offset += n;
                    // Finish on ZLP or non-full buffer
                    finished = n == 0 || n != 64;
                    queue.submit(RequestBuffer::reuse(comp.data, transfer_size));
                }
                tracing::trace!("Read data: {:x?}", &buf[..offset]);
                Ok(offset)
            } else {
                let comp = self.bulk_in(endpoint, RequestBuffer::new(buf.len())).await;
                comp.status.map_err(io::Error::other)?;

                let n = comp.data.len();
                buf[..n].copy_from_slice(&comp.data);
                tracing::trace!("Read data: {:x?}", &buf[..n]);
                Ok(n)
            }
        };

        block_on(fut.or(async {
            Timer::after(timeout).await;
            Err(std::io::ErrorKind::TimedOut.into())
        }))
    }
}
```

There's no `unsafe` here, but because I'm not familiar with async maybe I'm accidentally reusing a buffer somewhere?

xobs, 2025-05-10:
One thing I'm seeing is that `probe-rs` is on v0.1.12. Since then you've released v0.1.13, and there's been a large refactor on the road to v2.0.0.

I think I refactored my code correctly. It now looks like:

```rust
    fn read_bulk(&self, endpoint: u8, buf: &mut [u8], timeout: Duration) -> io::Result<usize> {
        let fut = async {
            let mut ep_in = self.endpoint::<Bulk, In>(endpoint).unwrap();
            let transfer = ep_in.allocate(buf.len());
            ep_in.submit(transfer);
            let Some(comp) = ep_in.wait_next_complete(timeout) else {
                return Err(std::io::ErrorKind::TimedOut.into());
            };
            comp.status.map_err(io::Error::other)?;

            let n = comp.actual_len;
            // If we got some number of bytes that is divisible by
            // wMaxPacketSize (and is nonzero), then we'll need to read a
            // zero length packet.
            if n != 0 && n % 64 == 0 {
                ep_in.submit(ep_in.allocate(0));
                ep_in.wait_next_complete(Duration::from_secs(1)).unwrap();
            }
            buf[..n].copy_from_slice(&comp.buffer);
            tracing::trace!("Read data: {:x?}", &buf[..n]);
            Ok(n)
        };

        block_on(fut.or(async {
            Timer::after(timeout).await;
            Err(std::io::ErrorKind::TimedOut.into())
        }))
    }
```

This doesn't seem to crash anymore, although I did have to add something to read the zero length packet.

1. Should we try to figure out why it crashes in v1.0.13, or is that effectively deprecated now?
2. Will the ZLP be handled before v2.0.0 is released? Or is that considered a non-issue?

kevinmehall, 2025-05-10:
The stack trace shows the crash occurring when cancelling transfers inside of `Queue::drop` as called from `InterfaceExt::read_bulk`. While the related nusb code has changed in v0.2, your new code does not drop the Endpoint with transfers pending so wouldn't trigger it. So I do want to investigate further and confirm it's not still possible on v0.2. Could you send a log including the crash? Set `RUST_LOG=nusb=debug` with `env_logger` or `tracing_subscriber`.

When your first version of `read_bulk` sets `n_transfers = buf.len() / 64` but also `transfer_size = buf.len()`, you are requesting reads totaling `n_transfers  * transfer_size == buf.len() * buf.len() / 64`, which is far larger than you are intending. The first transfer would read all the data. The second transfer would potentially receive the ZLP (explaining the difference in ZLP behavior you noted). Any remaining transfers would be cancelled when dropping the Queue when the function returns, which is where the crash occurs.  If the device sends further data without being prompted by another command, it might be consumed by those reads, and lost when it is never read from the `Queue`. These are all safe APIs and therefore the crash is definitely a bug in nusb, but it's presumably triggered by that atypical usage.

kevinmehall, 2025-05-10:
Is the ZLP issue when the device sends exactly `buf.len()` bytes followed by a ZLP? Your `n != 0 && n % 64 == 0 ` in both versions doesn't seem right -- it's only when the transfer completes with the full requested length that the ZLP hasn't been consumed. The difference for that case is explained by the extra reads your old code was submitting consuming that ZLP. 

The expected behavior is that a transfer completes after receiving a packet of less than max_packet_size bytes (a short or zero-length packet), or when its transfer length is reached. After the transfer length is reached, it does not wait for a subsequent packet. It doesn't know that you are expecting a ZLP there vs the data stream continuing into another transfer. Given that this function requires allocating a buffer anyway, the simplest way to deal with this might be to pad the buffer and requested transfer length with an additional `max_packet_size` bytes such that the transfer length is not reached unless the device sends more data than expected.

kevinmehall, 2025-05-10:
I believe I found the issue and fixed it in #130. Can you confirm that fix works for your code?

v0.2 is indeed not affected.

xobs, 2025-05-11:
That does seem to work for this particular implementation. And I've taken your feedback into the implementation, so it shouldn't be allocating extra data.

As for the ZLP -- I see how it could have accidentally worked, though there would be a dangling request. I'll fix that as well.

Thanks for the quick fix! v0.1.14 does fix the issue, so I'll close this.

closed 2025-05-11 by xobs