kevinmehall/nusb #149: [question] The interface is not loaded in time when hot plugging the device

LinusChen-yf, 2025-06-16:
When hot-plugging is detected, sometimes the device interfaces are empty, and it is often necessary to wait for 50 to 100 milliseconds to retry to reacquire the interface.
I think this kind of delay based on experience is not a good way. Is this a bug? Or is there any better way to deal with this problem? Thank you!

kevinmehall, 2025-06-16:
It's a [documented limitation](https://docs.rs/nusb/latest/nusb/fn.watch_devices.html#platform-specific-notes), but could be considered a bug that should be fixed if we can find a better way to do this.

This only occurs for a composite device using usbccgp.sys. We get the notification from the Windows API for the new "Composite Group Parent" device. But the interfaces are considered entirely separate devices by Windows with separate drivers that are created later.

We could potentially populate `DeviceInfo::interfaces()` using the interfaces from the configuration descriptor, though it would be missing the interface strings. But you still wouldn't be able to claim the interface until the Windows device nodes for the interfaces exist. And it would be hard to wait for them because they don't necessarily directly correspond to the interfaces in the interface descriptor.

If your device is entirely accessed by nusb and don't need to use other drivers on other interfaces, one workaround is to bind the entire device to WinUSB, and then there is a single Windows device node that is ready when the notification is received. Note that v0.1 has an issue where interfaces other than the first one don't work in this setup; fixed in v0.2 and pending backport in https://github.com/kevinmehall/nusb/pull/145.

kevinmehall, 2025-07-10:
That's fine. You can even `.open()` the device and just delay or retry the `.claim_interface()`.

`DeviceInfo::open()` only uses the ID, not any of the interface related information from the `DeviceInfo`.