kevinmehall/nusb #151: My `Interface::control_out` calls stopped working when upgrading to v0.2.0-beta.2

mfreeborn, 2025-07-03:
I was refactoring my code and thought it would be a nice opportunity to try the latest revision of `nusb`.

With minimal changes in the code to load my USB device based on an XMOS microcontroller, I'm able to open the device and claim the interface. However, when I submit a `Device::control_out` call, my device's endpoint 0 thread hangs.

From the perspective of the USB device, I can see that the `control` request gets all the way to the correct handler, i.e. every field in `ControlOut` has been received correctly, but then for some reason the endpoint 0 thread gets stuck in [this library function](https://github.com/xmos/lib_xud/blob/6f926f231d0c2a7067278be0d512ae4277d1b59b/lib_xud/src/user/client/XUD_EpFunctions.c#L171), (specifically in [this internal function](https://github.com/xmos/lib_xud/blob/6f926f231d0c2a7067278be0d512ae4277d1b59b/lib_xud/src/user/client/XUD_EpFunctions.c#L310)), which is to be called at the end of a Control Out request. I note that the linked library is provided by the manufacturer of the microcontroller, and the docs say that the function does this:

```
This function sends an empty packet back on the next IN request with PID1. It is
normally used by Endpoint 0 to acknowledge success of a control transfer.
```

I'm not entirely sure where the fault lies, other than to say the exact same `ControlOut` to the exact same `Interface` worked on v0.1.13, but doesn't on v0.2.0-beta.2.

Reading the following paragraph: 

```
Once the device descriptor is sent, a status transaction follows. If the transactions were successful, the host will send a zero length packet indicating the overall transaction was successful. The function then replies to this zero length packet indicating its status.
```

which one can find at the end of the `Control Transfers: The Bigger Picture` section on [this page](https://www.beyondlogic.org/usbnutshell/usb4.shtml#Control), I wonder if my device is stuck waiting for the host to send that zero-length packet to acknowledge the successful control transfer?

kevinmehall, 2025-07-03:
The details of the status stage ACK are handled by the OS and not nusb, so the question is what nusb is doing differently to make the OS behave differently.

Were you using the blocking or async API for the control transfer on v0.1? On Linux, v0.1 `control_out_blocking` uses a blocking ioctl, but this had the drawback of taking a device-wide lock in the kernel for the entire duration of the transfer. v0.2 uses the async ioctls for both async and blocking APIs (like libusb). I wouldn't expect these to behave any differently, but it is a difference.

The other difference in control transfers between v0.1 and v0.2 is that in v0.1, dropping the transfer future would cancel the transfer, while on v0.2 it has a timeout but does not otherwise cancel. Is it possible that the device's control OUT handler is taking too long, and the transfer is timing out before reaching the status stage? Many devices handle control transfer cancellation poorly and I could see that causing it to get stuck in a place like that.

You said "every field in ControlOut has been received correctly" but can you also confirm that the `wLength` in the setup packet as seen by the device is what you expect and the same between the two versions? I don't know how that would end up wrong, but if the host is expecting to send more data it wouldn't reach the status stage.

kevinmehall, 2025-07-03:
When a control transfer is cancelled, the device doesn't get any notification. This tends to be a problem for device-side APIs that involve blocking waiting for the control transfer to complete, while async or state-machine style APIs just sit there until the next control transfer.

The USB spec says that a device needs to be ready to accept a new SETUP packet at any time, and abort any pending previous control transfer. So if the device is stuck in state that another SETUP packet from a new control transfer doesn't break it out of, that should be considered a bug in the device-side USB stack.